adafruit/Adafruit_CircuitPython_MotorKit #11: board not supported

dakhunter, 2018-12-30:
I tried to import adafruit_motorkit in python3
the compiler throws back the following errors:
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import adafruit_motorkit
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/usr/local/lib/python3.5/dist-packages/adafruit_circuitpython_motorkit-1.2.4.dev0+g6f2d8ad.d20181230-py3.5.egg/adafruit_motorkit.py", line 59, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/usr/local/lib/python3.5/dist-packages/Adafruit_Blinka-1.1.0-py3.5.egg/board.py", line 65, in <module>
NotImplementedError: Board not supported
I have a Raspberry Pi 3 B model. As stated in the adafruit_blinka/board.py, this board is supported. This RPi worked before with the adafruit_motorkit library but for some reason it's not working now when I reinstalled the Operating system
Any help would be great

ladyada, 2018-12-30:
can you run this script and paste the output?

https://github.com/adafruit/Adafruit_Python_PlatformDetect/blob/master/bin/detect.py

dakhunter, 2018-12-30:
Here's the response from my RPi
```
python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import adafruit_platformdetect
i?", detector.board.any_raspberry_pi_40_pin)
print("Is this a BBB?", detector.board.BEAGLEBONE_BLACK)
print("Is this an Orange Pi PC?", detector.board.ORANGE_PI_PC)
print("Is this an embedded Linux system?", detector.board.any_embedded_linux)
print("Is this a generic Linux PC?", detector.board.GENERIC_LINUX_PC)
```

```
if detector.board.any_raspberry_pi:
    print("Raspberry Pi detected.")>>> 
>>> detector = adafruit_platformdetect.Detector()
>>> 
>>> print("Chip id: ", detector.chip.id)
Chip id:  BCM2XXX
>>> 
>>> print("Board id: ", detector.board.id)
Board id:  None
>>> 
>>> print("Is this a Pi 3B+?", detector.board.RASPBERRY_PI_3B_PLUS)
Is this a Pi 3B+? False
>>> print("Is this a 40-pin Raspberry Pi?", detector.board.any_raspberry_pi_40_pin)
Is this a 40-pin Raspberry Pi? False
>>> print("Is this a BBB?", detector.board.BEAGLEBONE_BLACK)
Is this a BBB? False
>>> print("Is this an Orange Pi PC?", detector.board.ORANGE_PI_PC)
Is this an Orange Pi PC? False
>>> print("Is this an embedded Linux system?", detector.board.any_embedded_linux)
Is this an embedded Linux system? True
>>> print("Is this a generic Linux PC?", detector.board.GENERIC_LINUX_PC)
Is this a generic Linux PC? False
>>> 
>>> if detector.board.any_raspberry_pi:
...     print("Raspberry Pi detected.")
... 
Raspberry Pi detected.

```

ladyada, 2018-12-30:
ok - please paste the output of `cat /proc/cpuinfo`

dakhunter, 2018-12-30:
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: 2a02082
Serial		: 00000000f60f3209