adafruit/Adafruit_CircuitPython_MotorKit #14: I2C issue with BNO055

jfclifford, 2019-02-07:
Trying to connect BNO055 and Motor Shield v2.3 at the same time to a Metro M4 Express.  MotorKit appears to be claiming the i2c bus during init.  I would assume this would cause issues with other i2c devices as well, but I do not have any to test with at the moment.
`"""Simple test for using adafruit_motorkit with a DC motor"""
import time
import board
import busio
import adafruit_bno055
from adafruit_motorkit import MotorKit
 
i2c = busio.I2C(board.SCL, board.SDA)
sensor = adafruit_bno055.BNO055(i2c)

kit = MotorKit()
 
kit.motor1.throttle = 1.0
time.sleep(0.5)
kit.motor1.throttle = 0`

`code.py output:
Traceback (most recent call last):
  File "code.py", line 11, in <module>
  File "adafruit_motorkit.py", line 75, in __init__
ValueError: SCL in use`

ladyada, 2019-02-07:
ooh i though the main i2c was ‘static’? @tannewt @dhalbert

sommersoft, 2019-02-07:
@ladyada, it is. The issue is that the I2C object is created during instantiation [here](https://github.com/adafruit/Adafruit_CircuitPython_MotorKit/blob/master/adafruit_motorkit.py#L75).

I think the easiest way ahead is to allow the I2C object to be passed in as an option, so that MotorKit doesn't claim or fail otherwise.

caternuson, 2019-02-07:
That's what I was thinking too. Pass it in like anything else:
```python
 def __init__(self, i2c, address=0x60):
```
and delete:
https://github.com/adafruit/Adafruit_CircuitPython_MotorKit/blob/master/adafruit_motorkit.py#L75

sommersoft, 2019-02-07:
Yeah. Or keyword it and adjust accordingly:

tannewt, 2019-02-08:
I think using `board.I2C()` would be best. It's a singleton so both the library and the user code can call that and get the same object. For this high level of a library I wouldn't keyword arg it because it being different may imply other differences.

caternuson, 2019-02-08:
Neat! Forgot about those board level defines.

eudoxos, 2019-02-23:
I am seeing the same issue, trying to use SSD1306 and stepper FeatherWing on a *Feather M4 Express with samd51j19* (with *CircuitPython 4.0.0-beta.2 on 2019-02-05*). The second device to be initialized (tried both ways) raises the `ValueError: SCL in use` exception. Something I can do about that?

```python
import busio, board, adafruit_ssd1306
i2c=busio.I2C(board.SCL, board.SDA)
oled=adafruit_ssd1306.SSD1306_I2C(128,32,i2c)

import adafruit_motorkit
from adafruit_motor import stepper
mkit=adafruit_motorkit.MotorKit()

```

dhalbert, 2019-03-17:
@ladyada and @dhalbert discussed, and would do:

```
def __init__(self, address=0x60, i2c=None):
    if i2c is None:
        i2c = board.I2C()
```

ladyada, 2019-03-19:
Hiya this should be fixed now - please try the latest version, and if you're still having issues re-open the ticket :)

closed 2019-03-19 by ladyada