adafruit/Adafruit_CircuitPython_MotorKit #29: Odroid-C2 w/ Armbian Buster (current) issues

crazyquark, 2020-03-19:
Hello!

I've used the previous version of this with an Odroid-C1+ but decided to upgrade to the latest version, following this tutorial: [tutorial](https://learn.adafruit.com/circuitpython-libaries-linux-odroid-c2/initial-setup)
Sadly, there a few issues, hope you can guide me in the right direction:

- Libgpiod cannot be installed by compiling, but there is a python3-libgpiod in the repos which I installed(it complains about kernel headers, they added a check in the compilation scripts, installing the meson64 kernel headers does not help, armbian is weird about kernels); this is somewhat fixed let's say by installing the pre-packaged one
- SPI is not enabled in the kernel, needs kernel recompiling, not doing that; reported on armbian forums
- permissions for /dev/gpiodevice* need to be set from udev rules our you could always run as root
- The real issue: my python3 interpreter freezes at times after setting the throttle to one of the motors to 0.5-1. I even got a kernel oops at one point. The odroid is fine but the python3 interpreter just freezes, and the motor keeps going until I cut power. It sometimes unfreezes after a while. I now power the motors and the C2 independently to rule out any power supply noise. Any tips would be appreciated, I can debug, just tell me how.

My armbian logs:
[logs](http://ix.io/2eFk)

ladyada, 2020-03-19:
what if you remove the motors from the shield and run the script?

makermelissa, 2020-03-19:
Can you verify that you're installing the libgpiod2 package as well? python3-libgpiod is only the python3 bindings, but you need the library as well if you're installing it via packages.

For SPI, we couldn't get it working either, which is why it says SPI isn't enabled on https://learn.adafruit.com/circuitpython-libaries-linux-odroid-c2/faq-troubleshooting-2.

crazyquark, 2020-03-19:
@ladyada yes, if the motors are not connected it does not freeze, please note that I am now powering the motors separately but who knows... 
@makermelissa yes, the libgpiod2 is installed as dependency of python3-libpiod
Actually, I was able to get SPI working by switching to nigthly builds and editing the DTB... quite hackish, see this: https://forum.armbian.com/topic/13389-crazyquark/?tab=comments#comment-97282

ladyada, 2020-03-19:
if removing the motors stops the problem... its almost certainly motor noise :) it can feed thru the ground. add caps and keep the motors FAR AWAY :)
https://www.pololu.com/docs/0J15/9
https://medium.com/jungletronics/dc-motors-against-back-emf-589d8ed174cc

crazyquark, 2020-03-19:
OK, on the nightly kernel, I can;t get it to freeze but I now get this exception at times:
kit.motor2.throttle = 0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/chris/.local/lib/python3.7/site-packages/adafruit_motor/motor.py", line 73, in throttle
    self._positive.duty_cycle = 0xffff
  File "/home/chris/.local/lib/python3.7/site-packages/adafruit_pca9685.py", line 92, in duty_cycle
    self._pca.pwm_regs[self._index] = (0x1000, 0)
  File "/home/chris/.local/lib/python3.7/site-packages/adafruit_register/i2c_struct_array.py", line 74, in __setitem__
    i2c.write(buf)
  File "/home/chris/.local/lib/python3.7/site-packages/adafruit_bus_device/i2c_device.py", line 103, in write
    self.i2c.writeto(self.device_address, buf, start=start, end=end, stop=stop)
  File "/home/chris/.local/lib/python3.7/site-packages/busio.py", line 86, in writeto
    return self._i2c.writeto(address, memoryview(buffer)[start:end], stop=stop)
  File "/home/chris/.local/lib/python3.7/site-packages/adafruit_blinka/microcontroller/generic_linux/i2c.py", line 38, in writeto
    self._i2c_bus.write_bytes(address, buffer[start:end])
  File "/home/chris/.local/lib/python3.7/site-packages/Adafruit_PureIO/smbus.py", line 256, in write_bytes
    self._device.write(buf)
And the command does not execute

ladyada, 2020-03-19:
whats the *actual error* , its been cut off

crazyquark, 2020-03-19:
Ah, sorry, it's:
OSError: [Errno 6] No such device or address

ladyada, 2020-03-19:
please hard reboot, something went bad in the i2c. 
as we think this is all wiring issues, not code at all, it would be better if you posted in the tech support forums than here because it is not a code issue and our engineers can assist :)
please re-open if you can show it is a software bug :)

closed 2020-03-19 by ladyada

crazyquark, 2020-03-22:
FYI, I just tested, same HAT, same code, on a RPi 3 it works flawlessly, so I am not convinced it is a wiring or hardware issue, I am pretty sure it's a software one. It could be that the i2c driver in Armbian is just unreliable.
The odd thing is that only crashes when the motors are connected and/or powered.