adafruit/Adafruit_CircuitPython_MotorKit #32: Stepper Motor Speed Limitations

Xeonzinc, 2020-09-27:
Hi, 

I have the raspberry PI hat, and a couple of stepper motors. I am able to use this library ok, but am running into issues with the speed of which I can drive the motors using the oneStep function.

I was under the impression the purpose of the hat was to offload the repetitive processing of the stepper motors onto a separate discrete board to avoid interruptions you would get on the main program/PI, but the fact that I have to call oneStep from python on the raspberry PI seems to go against this purpose and (I believe) leads to my code/PI being the limiting factor in the speed the stepper motors can run, as every single step has to be called at separately, and at exactly the correct time.

In other stepper libraries you seem to offer multistep functions [here](https://learn.adafruit.com/adafruit-motor-shield/using-stepper-motors) . I'm assuming this command goes direct to the hat and avoids any other program interruptions for the duration of the command (although could be wrong). Is there anywhere to replicate this functionality within this library/ for the PI hat?

---------

I am aware I could get a motor with different gearing to change the speed, but when then likely end up sacrificing on turn resolution. And this code approach seems like a fundamental issue.

Xeonzinc, 2020-09-27:
Just to add a little more detail with one stepper I can achieve 8RPM on a PI-zero and 22RPM on a PI-4 with identical code. Indicating it is the python code which is the bottle neck on the PI-zero, but equally on the PI-4 if I had other threads running (in any kind of real application) this will inevitably reduce the maximum RPM there too. So having a method to send multiple steps to the hat at once would be beneficial.

However, having looked at the motor stepper class it seems like the individual coil activation is done within the python thread, I may have misunderstood the hats purpose. If someone can confirm the hat is only for digital control of power management rather than coil sequence activation logic I will work off this limitation and build the required overhead into my code. Thanks

knoxvillesjoker, 2020-11-27:
I think it is a limitation of the way the pca9685 chipset initiates movement to the stepper motor.  I am not sure if a library update could work as if it were possible to send it direct positional data, things might be a bit faster instead of having to use for loops.