adafruit/Adafruit_CircuitPython_MotorKit #36: Stepper motor vibrates violently

zwying0814, 2020-12-20:
There is a problem that puzzles me. When I use a stepper motor, following the example program, the motor vibrates violently.
This is my code：

```
import time
from adafruit_motorkit import MotorKit
kit = MotorKit()
for i in range(100):
    kit.stepper1.onestep()
    time.sleep(0.01)
kit.stepper1.release()
```

ladyada, 2020-12-20:
hi i think you have a wiring issue, please post in your vendor tech support forums - this isnt a library issue :)

closed 2020-12-20 by ladyada

petrasvestartas, 2023-07-16:
Hi,

After being frustrated for some time, I stopped the given battery pack and more importantly jumper wires + circular connector. There was a lot of vibration, noise.

I finally bought 12 V 600mA power supply.

This was not enough... Then there was a power supply with jumper wires going from adafruit board with a round connector. I demounted it and stopped using it. Looks simply, easy to screw, but seems to not deliver all the power or something gets lost along the way. 

At the end, I took 12 V 600A power supply. Then cut the connector with scissors to get access to normal copper wires.
And directly then screwed to the Adafruit board. Seems to work much better and less noise.

**I wish, that a person who has a proper electronics background could explain why this happened.**

In short summary these photos illustrates everything:

[image]

dhalbert, 2023-07-16:
@petrasvestartas Could you bring this up (with same content) in https://forums.adafruit.com, maybe https://forums.adafruit.com/viewforum.php?f=8 ? You can get some help from our general electronics support folks.