adafruit/Adafruit_CircuitPython_MotorKit #45: Using motorkit with Adafruit DC Motor Bonnet on Jetson Nano

littlebee, 2022-04-18:
Hi!   I'm trying to get the [Adafruit DC & Stepper Motor Bonnet](https://www.adafruit.com/product/4280) working on a Nvidia Jetson Nano 4Gb using MotorKit.

I followed the instructions for setting up [Circuit Python on the Nano](https://learn.adafruit.com/circuitpython-libraries-on-linux-and-the-nvidia-jetson-nano/initial-setup).  The motor board is connected to the directly to the Nano and powered sepately from the Nano.

I2C shows the two I2C addresses:
```
$i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: 60 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --
```

When I try to `from adafruit_motorkit import MotorKit` in python3 console, however, I get an error:

```
>>> from adafruit_motorkit import MotorKit
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/bee/.local/lib/python3.7/site-packages/adafruit_motorkit.py", line 42, in <module>
    import adafruit_motor.motor
  File "/home/bee/.local/lib/python3.7/site-packages/adafruit_motor/motor.py", line 26, in <module>
    from pwmio import PWMOut
  File "/home/bee/.local/lib/python3.7/site-packages/pwmio.py", line 48, in <module>
    raise NotImplementedError("pwmio not supported for this board.")
NotImplementedError: pwmio not supported for this board.
```

I additionally tried enabling the PWM pins using jetson-io.py, but that didn't fix the error.   My jetson GPIO config looks like:
```
|                Select desired functions (for pins):                |
 |                                                                    |
 |                 [ ] aud_mclk      (7)                              |
 |                 [ ] i2s4               (12,35,38,40)                    |
 |                 [*] pwm0           (32)                             |
 |                 [*] pwm2           (33)                             |
 |                 [*] spi1              (19,21,23,24,26)                 |
 |                 [ ] spi2              (13,16,18,22,37)                 |
 |                 [ ] uartb-cts/rts (11,36)                          |
 |                                                                    |
 |                                Back                                |
```
 
 Thank you for any help!

tekktrik, 2022-04-20:
PWM support may be available on the Jetson Nano based on my brief look at Blinka, @makermelissa is that correct?  If so, then it seems like because that library relies on `pwmio` you wouldn't be able to use it on that board. :/

tekktrik, 2022-04-20:
Oh, but they're only used for type annotations, so I guess that's what's causing the issue in the Blinka environment.

FoamyGuy, 2022-04-20:
I don't have any experience with pwm on Blinka and have never used a jetson nano, so I can't add much to that discussion. 

I do think a protocol makes sense if that import is only used for typing though. My first instinct would be to put it into `circuitpython_typing` instead of Blinka directly. iirc the type helpers were moved from Blinka into the `circuitpython_typing` library so that they could be used by Blinka and in other places needed like this library.

tekktrik, 2022-04-20:
Okay, I think adding a `typing.Protocol` there that could actually be imported for typing in a Blinka environment should resolve this.  I can get on that.

tekktrik, 2022-04-20:
Whoopsie!  Didn't mean to close

tekktrik, 2022-04-21:
Opened an issue for adding the protocol.  Once that's merged (or the decision is made to move the protocol into `adafruit_motor`) I can write a fix for this!

closed 2022-04-21 by dhalbert

tekktrik, 2022-04-22:
HI @littlebee, if you upgrade `adafruit_motor` to 3.4.0 (now on PyPI), it should fix your problem!

littlebee, 2022-04-23:
Thank you!!!  That totally works.  So glad not to have to switch boards!