adafruit/Adafruit_CircuitPython_MotorKit #48: adafruit-circuitpython-motorkit to support Raspberry PI5

SlavPetkovic, 2024-01-15:
This library with example code works perfectly fine on RPi4 however when i run it on RPi5 with latest Bookworm-32 or 64 regardless I am getting issues while using Adafruit DC & Stepper Motor HAT.
Issue: Cannot determine SOC peripheral base address

After some research I found that RPi.GPIO, wiringPI, and pigpio do not work on the Pi5 which has new hardware for the GPIO.

Link to rpi forum with discussion on this issue:
[https://forums.raspberrypi.com/viewtopi ... 8#p2160578](https://forums.raspberrypi.com/viewtopic.php?p=2160578#p2160578)

After following instructions:

```
sudo apt remove python3-rpi.gpio
pip3 install rpi-lgpio
```
Different error appearead:
  File "/home/slav/Projects/MarsRover/rover/lib/python3.11/site-packages/adafruit_motor/motor.py", line 63, in DCMotor
    def __init__(self, positive_pwm: PWMOut, negative_pwm: PWMOut) -> None:
                                     ^^^^^^
NameError: name 'PWMOut' is not defined

I think  that the Adafruit CircuitPython Motorkit library needs modifications to work with a Pi5.|

makermelissa, 2024-03-18:
I think fixing https://github.com/adafruit/Adafruit_Blinka/issues/776 should resolve this issue.

makermelissa, 2024-03-19:
PWM support now added for the Pi 5. Please try again with the latest version of Blinka.

makermelissa, 2024-03-20:
Tested with hardware and verified it now works.

closed 2024-03-20 by makermelissa

SlavPetkovic, 2024-03-20:
Thank you, Melissa, for working on this. I am excited to give it a try as well.